cpq/bare-metal-programming-guide #25: suggested fix

uliano, 2023-01-30:
the last line of clock_init

while ((RCC->CFGR & 12) == 0) spin(1);              // Wait until done

should probably be:

while ((RCC->CFGR & 12) != 8) spin(1);

uliano, 2023-01-30:
@cpq Hi Sergey,

thanks for coming back. Yes I was meaning exactly that, it is a “fix” for something that’s already working :-D!

But many more thanks for the effort you put in your tutorial, it provided me a "way out” from AVRs the only MCU I was able to program bare metal (and design hardware too). 

In a couple of days I ported your code (without the web server, for sure) to a blackpill F411 and I learned a lot!

My next endeavor will be irq on serial monitor or maybe even exploiting the USB connector of the board

cpq, 2023-01-30:
@uliano awesome, that's great to hear.

You might want to check out this: https://github.com/cesanta/mongoose/tree/master/examples/stm32/nucleo-f429zi-usb-rndis

This is a bare metal firmware for F429. I think it'll work on 411.
It should make your F411 to look like an Ethernet connected device (ethernet-over-USB)
It uses our Mongoose Library of course.
With that approach, you can talk to your F411 over a usual browser / REST API.

closed 2024-11-15 by cpq