cpq/bare-metal-programming-guide #28: Webserver code is handling VLAN packets

rbaker83, 2023-02-27:
I recently built the step-7 web server project and connected it to my home network.  My hard wired network is running through a TP-Link vlan switch.  I am running on a Nucleo F429-ZI board.  The log messages below appear to indicate that the parser is not handling VLAN tagged frames (last line below) properly.  This is not an area of expertise for me and I am wondering if there is a setting I need to tweak or some change in the code that i can make to get around this issue?  I am new to this so please forgive me if this is not the proper method to ask a question.  

Thank you
Rick Baker

1c914 3 mongoose.c:6904:tx_dhcp_discove DHCP discover sent
1c91a 3 mongoose.c:3474:mg_connect      350 0xffffffff udp://time.google.com:123
1c921 3 mongoose.c:3474:mg_connect      351 0xffffffff udp://8.8.8.8:53
1c927 3 mongoose.c:7406:mg_connect_reso 351 00000000:49268->08080808:53
1c92d 1 mongoose.c:404:mg_error         351 0xffffffff net down
1c933 1 mongoose.c:404:mg_error         351 0xffffffff DNS send
1c938 3 mongoose.c:3474:mg_connect      352 0xffffffff mqtt://broker.hivemq.com:1883
1c940 1 mongoose.c:404:mg_error         351 0xffffffff net down
1c945 1 mongoose.c:404:mg_error         351 0xffffffff DNS send
1c94b 1 mongoose.c:404:mg_error         352 0xffffffff net down
1c950 1 mongoose.c:404:mg_error         352 0xffffffff net down
1c956 1 mongoose.c:404:mg_error         352 0xffffffff net down
1c95b 1 mongoose.c:404:mg_error         352 0xffffffff net down
1c961 1 mongoose.c:404:mg_error         352 0xffffffff net down
1c967 3 mongoose.c:3451:mg_close_conn   352 0xffffffff closed
1c96c 1 mongoose.c:404:mg_error         350 0xffffffff DNS error
1c972 3 mongoose.c:3451:mg_close_conn   351 0xffffffff closed
1c977 3 mongoose.c:3451:mg_close_conn   350 0xffffffff closed
1c97c 3 mongoose.c:7278:mip_rx            Unknown eth type 8899

cpq, 2023-02-28:
@rbaker83 a legit question - and yeah, currently, the built-in TCP/IP stack does not handle vlan-tagged frames!
A fix is coming soon.

rbaker83, 2023-03-01:
Error not caused by loop detect packets. I am not sure why it began working but all us okay now.

closed 2023-03-01 by rbaker83