cpq/bare-metal-programming-guide #38: build problem for stm32f411re

dimitri1949, 2023-12-05:
Hi!
I have Nucleo-64 stm32f411re and MikroelEktronica+ stm32f407zg boards,
adapting example code to f411re and f407zg worked perfectly with 2023 February version with manually added include directory and startup.c (modified obviously).

Current examples with SOURCES += cmsis_f4/Source/Templates/gcc/startup_stm32f411xe.s do not build 
with errors:
in func. LoopFillZerobss in startup_stm32f411xe.s -- undefined reference to SysInit
in __libc_init_array.. -- undefined reference to _init
It seems that some path is now broken since startup_stm32f411xe.s and startup_stm32f429xx.s look very similar.
[image]

Also copying 200MB of files every time when new version of a project is created seems inefficient.
I copied cmsis_core and cmsis_f4 directories (dowdloaded by git clone..) to d:\gcc-arm\Work\ directory, but how to modify Makefile to access them is a mystery for me, pleaser help.

Best regards
dimitri1949

gillesxr, 2023-12-07:
Hello!

A fast fix is to add a blank _init function in the main.c file.
`void _init(void) {}`

To access to CMSIS from a Makefile:
Create a variable in the Makefile, i.e. `CMSIS_CORE_PATH=d:\gcc-arm\Work\cmsis_core`, and use it with `$(CMSIS_CORE_PATH)`.

BR
gillesxr.

dimitri1949, 2023-12-07:
Hello!

void _init(void) {} was actually sitting in the end of syscalls.c file!

I was careless!

I modified step-5-cmsis example project to stm32f411re from scratch,
essentially changing only:
hal.h:
#include "stm32f429xx.h" -> stm32f411xe.h
UART3 -> UART6

main.c:
led -> LED_PA5

link.ld:
MEMORY
{ /* stm32f411re */
flash(rx) : ORIGIN = 0x08000000, LENGTH = 512K
sram(rwx) : ORIGIN = 0x20000000, LENGTH = 128K
}

MakeFile:
SOURCES += cmsis_f4/Source/Templates/gcc/startup_stm32f429xx.s ->
SOURCES += cmsis_f4/Source/Templates/gcc/startup_stm32f411xe.s

and now it works!
So all problems where my fault!

I will now try it with cmsis_core and cmsis_f4 in d:\gcc-arm\Work\

BR
Dimitri

closed 2024-11-10 by cpq