cpq/bare-metal-programming-guide #57: clock init from HSE > no network activity from mongoose network lib

seganoff, 2025-06-10:
Hi everyone,

im struggling for weeks now trying to make the step7 runable, but instead of using HSI(RC) like in the whole tutorial i wanna use HSE(dedicated xtal).
im using the 767Nucleo board: 8 MHz hse comes from on board stlink (reflashed into jlink).

The issue is: as soon as i use modified clock init routine, there is no network activity anymore.
From router perspective: Tx frames are send out, but Rx frames counter simply 0

Now, im not sure what is the issue here, whats causing this odd behaviour. 
Is it my clock_init or mongoose simply cannot run when HSE used?

Are there any examples how to use mongose & where PLL is fed by HSE? I have not encountered such...

please have a look,
https://github.com/seganoff/mcu/commit/914455f9d05cc3f70f47544eea9c6f5295e48c56
& the parent commit, which is simply unzip wizard.zip (board=f767&ide=GCC+make&rtos=baremetal)
clock_init is defined in hal.h:GG(file end) & calls happen in hal.c:160

you can also look here, but this is like a dev branch, no clean commit history
https://github.com/seganoff/f401ccu6/tree/cpq/bare-metal-programming-guide/gh_cpqstep6

thanx in advance, any help welcomed & appreciated

cpq, 2025-06-11:
@seganoff thanks

Does Ethernet IRQ handler gets triggered at all?

What is the CPU frequency?

Did you try to change the [frequency divider](https://github.com/cesanta/mongoose/blob/9411f1c7b6667e3f7c52fde50fb1eac4388e7e20/src/drivers/stm32f.h#L18) in the driver? To change the divider, just define this in your mongoose_config.h:

```c
#define MG_DRIVER_MDC_CR XXX
```

seganoff, 2025-06-11:
$ picocom /dev/ttyACM0 -b 115200

HSE run

```
0      2 main.c:13:main                 HAL initialised, starting firmware...
7      2 mongoose.c:21818:mg_phy_init   PHY ID: 0x07 0xc131 (LAN87x)
d      2 mongoose.c:4154:mg_mgr_init    Driver: stm32f, MAC: 02:29:c0:04:02:32
14     2 mongoose_impl.c:901:mongoose_i Starting HTTP listener
19     3 mongoose.c:4078:mg_listen      1 0 http://0.0.0.0:80
1f     2 mongoose_impl.c:943:mongoose_i Mongoose init complete
24     1 mongoose.c:5132:mg_tcpip_poll  Network is down
3f5    1 mongoose.c:5132:mg_tcpip_poll  Network is down
7dd    3 mongoose.c:23309:mg_tcpip_driv Link is 100M full-duplex
7e2    1 mongoose.c:4381:onstatechange  Link up
7e7    3 mongoose.c:4491:tx_dhcp_discov DHCP discover sent. Our MAC: 02:29:c0:04:02:32
bc5    3 mongoose.c:4491:tx_dhcp_discov DHCP discover sent. Our MAC: 02:29:c0:04:02:32
fad    3 mongoose.c:4491:tx_dhcp_discov DHCP discover sent. Our MAC: 02:29:c0:04:02:32
1395   3 mongoose.c:4491:tx_dhcp_discov DHCP discover sent. Our MAC: 02:29:c0:04:02:32
177d   1 mongoose.c:23262:mg_tcpip_driv No free descriptors
1782   3 mongoose.c:4491:tx_dhcp_discov DHCP discover sent. Our MAC: 02:29:c0:04:02:32
1b65   1 mongoose.c:23262:mg_tcpip_driv No free descriptors
1b6a   3 mongoose.c:4491:tx_dhcp_discov DHCP discover sent. Our MAC: 02:29:c0:04:02:32
1f4d   1 mongoose.c:23262:mg_tcpip_driv No free descriptors
1f52   3 mongoose.c:4491:tx_dhcp_discov DHCP discover sent. Our MAC: 02:29:c0:04:02:32
2335   1 mongoose.c:23262:mg_tcpip_driv No free descriptors
233a   3 mongoose.c:4491:tx_dhcp_discov DHCP discover sent. Our MAC: 02:29:c0:04:02:32
271d   1 mongoose.c:23262:mg_tcpip_driv No free descriptors
2722   3 mongoose.c:4491:tx_dhcp_discov DHCP discover sent. Our MAC: 02:29:c0:04:02:32
2b05   1 mongoose.c:23262:mg_tcpip_driv No free descriptors
...
```
HSI run
```
0      2 main.c:13:main                 HAL initialised, starting firmware...
7      2 mongoose.c:21818:mg_phy_init   PHY ID: 0x07 0xc131 (LAN87x)
d      2 mongoose.c:4154:mg_mgr_init    Driver: stm32f, MAC: 02:29:c0:04:02:32
14     2 mongoose_impl.c:901:mongoose_i Starting HTTP listener
1a     3 mongoose.c:4078:mg_listen      1 0 http://0.0.0.0:80
1f     2 mongoose_impl.c:943:mongoose_i Mongoose init complete
25     1 mongoose.c:5132:mg_tcpip_poll  Network is down
3f5    1 mongoose.c:5132:mg_tcpip_poll  Network is down
7dd    1 mongoose.c:5132:mg_tcpip_poll  Network is down
bc5    1 mongoose.c:5132:mg_tcpip_poll  Network is down
fad    1 mongoose.c:5132:mg_tcpip_poll  Network is down
1395   1 mongoose.c:5132:mg_tcpip_poll  Network is down
59e5   1 mongoose.c:5132:mg_tcpip_poll  Network is down
[...] hook up lan cable...
5dcd   3 mongoose.c:23309:mg_tcpip_driv Link is 100M full-duplex
5dd2   1 mongoose.c:4381:onstatechange  Link up
5dd7   3 mongoose.c:4491:tx_dhcp_discov DHCP discover sent. Our MAC: 02:29:c0:04:02:32
61b5   3 mongoose.c:4491:tx_dhcp_discov DHCP discover sent. Our MAC: 02:29:c0:04:02:32
659d   3 mongoose.c:4491:tx_dhcp_discov DHCP discover sent. Our MAC: 02:29:c0:04:02:32
6985   3 mongoose.c:4491:tx_dhcp_discov DHCP discover sent. Our MAC: 02:29:c0:04:02:32
6d6d   3 mongoose.c:4491:tx_dhcp_discov DHCP discover sent. Our MAC: 02:29:c0:04:02:32
7155   3 mongoose.c:4491:tx_dhcp_discov DHCP discover sent. Our MAC: 02:29:c0:04:02:32
753d   3 mongoose.c:4491:tx_dhcp_discov DHCP discover sent. Our MAC: 02:29:c0:04:02:32
7782   3 mongoose.c:4470:tx_dhcp_reques DHCP req sent
7787   2 mongoose.c:4609:rx_dhcp_client Lease: 86400 sec (86430)
778d   1 mongoose.c:4378:onstatechange  Got IP
7791   2 mongoose.c:4374:onstatechange  READY, IP: 192.168.3.245
7797   2 mongoose.c:4375:onstatechange         GW: 192.168.3.1
779d   2 mongoose.c:4376:onstatechange        MAC: 02:29:c0:04:02:32
89a6   3 mongoose.c:4527:rx_arp         ARP: tell 192.168.3.1 we're 02:29:c0:04:02:32
```

seganoff, 2025-06-11:
lets uncomment & inspect the printf 
`printf("D0 %lx SR %lx\n", (long) s_txdesc[0][0], (long) ETH->DMASR);`
```
                     233c   3 mongoose.c:4491:tx_dhcp_discov DHCP discover sent. Our MAC: 02:29:c0:04:02:32
271d   1 mongoose.c:23262:mg_tcpip_driv No free descriptors
D0 b0100000 SR 680084
                     2724   3 mongoose.c:4491:tx_dhcp_discov DHCP discover sent. Our MAC: 02:29:c0:04:02:32
2b05   1 mongoose.c:23262:mg_tcpip_driv No free descriptors
D0 b0100000 SR 680084
```

seganoff, 2025-06-11:
this one brought me to right direction, 
https://stackoverflow.com/questions/50834140/stm32-eth-irqhandler-never-hit/79167700#79167700
& of course the request to check if ether_irqHandler gets triggered @cpq 

actually changing the sram defines in linkerscript solved the problem 
( well for me at the moment still incompehendable how what & why , my disgust for documentation style of st is just skyrocketting)

```
  4       /*sram1 368kb+sram2 16kb*/
  5   /*sram(rwx) : ORIGIN = 0x20020000, LENGTH = 384k*/
  6     sram(rwx) : ORIGIN = 0x20000000, LENGTH = 512k
  7 /*dtcm(rwx) : ORIGIN=  0x20000000, LENGTH = 128k*/
```

[image]

DTCM, same as CCM to put methods that need to be run fast...
from my understanding ram starts from 2002_0000, memory protection unit MPU was not turned on...

seganoff, 2025-06-14:
@cpq it all depends on employers wish to have that functionality implemented... 
Personally i was not considering this, maybe in some further future.
But there are huge departments who deal with such bring in procedures, from economic aspects to research & development